Man-fred/culfw.esp8266 #3: rf_moritz compile failed

kroon040, 2020-06-13:
Hi Manfred

I see that moritz you have enabled.

It fails when I compile

Multiple libraries were found for "WiFiUdp.h"
/home/richard/Arduino/libraries/rf_moritz/rf_moritz.c:5:27: fatal error: avr/interrupt.h: No such file or directory
 Used: /home/richard/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi
 #include <avr/interrupt.h>
 Not used: /home/richard/Downloads/arduino-1.8.12/libraries/WiFi

regards Richard

kroon040, 2020-06-13:
Manfred

I compiled the new version. Setting the network was a little difficult , dhcp didn't work, so static went ok.

I have now  4 thermostat basics 1 wallthermostat and 1 fakewallthermostat, and looks like it works

Good job. I will check tomorrow late if everything is stable

Man-fred, 2020-06-14:
Hi Richard,
great that it works for you, my MAX!-devices seems also running, so I can rollout it in all rooms.
Regards, Manfred

kroon040, 2020-06-15:
Manfred,

It looks stable. Only the the hostname (devicename) is not working. In the console commad Rid I get esp-cul, but on the network it shows still ESP-F44XXXX. Also reboot and/erased reflashed but not working.

regards Richard

Man-fred, 2020-06-16:
Hi,
the problem with the hostname is a problem in dns/dhcp. Perhaps it helps to delete the name in the dns-server and then restart the esp. Then it should show the new and correct name.

kroon040, 2020-06-18:
Hi,

Removed everything from dns/dhcp entries, erased the the esp, and reprogrammed. Tried both esps, and both gives ESP-F****

Man-fred, 2020-06-19:
Hi,
I will test it. I have two more esp and will upload the sketch. Maybe then I can see the problem myself.

Man-fred, 2020-06-21:
Good news: I found the error in setting the hostname. But i changed something more and must test if the changes are stable.

kroon040, 2020-06-22:
Good to hear. Let me know when I can test it.

My 4 thermostat basic and 1 wandthermostat works fine at the moment.

Man-fred, 2020-07-04:
Hi Richard,
I pushed the changes in testing the hostname to master. Nothing else is changed, so the stability shouldn't be affected. 
Regards, Manfred

kroon040, 2020-07-05:
Manfred,

Now it works. the hostname comes correct in the wifi router

thnx , I close this issue. Thnx for all the support.

closed 2020-07-05 by kroon040