Man-fred/culfw.esp8266 #12: CULW.8266 started with a false IP adress

romakrau, 2021-12-22:
Hello,
I flashed the bin file V01-67-00.clufw-esp8266.ino.d1_mini.bin. I add the Wis,Wik,WiD,Wio as named in the readme file. I got this ouput on the serial monitor:
`⸮⸮~⸮65432
UDP 1, TCP 0 on 192.168.178.244:2323

Channel 12 
CC1100_PARTNUM 0x00: FF
CC1100_VERSION 0x14: FF
`
It is NOT my IP adress. I tried to compile the source but it was not successful due to many libaries missing.
Could you be so kind and help me.
Thank you
Roman

romakrau, 2021-12-23:
Hallo Manfred,
ich habe es soweit hinbekommen dass ich mich zumindest verbinden mit dem 8266 verbinden kann. Hilfreich war hier das Studium des Wiki commendref. Der Compiler (Arduino 1.8.15) bleibt immer an der Stelle cc1100.h Zeile 33 `#ifdef ESP8266
	void assert(void);
	void deassert(void);
 #endif` hängen. Vielleicht kannst Du damit ja was anfangen undmir weiterhelfen.
Gruß Roman

Man-fred, 2021-12-23:
Hallo Roman,
ich komme jetzt zum Testen und schreib dann mal meine Ergebnisse.
Viele Grüße
Manfred

romakrau, 2021-12-23:
Hallo Manfred,
dank Dir für Deine schnelle Hilfe. Leider bekomme ich immer noch eine Fehlermeldeung (sieh Code).
`Arduino: 1.8.15 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/mmu_iram.h:22,

                 from C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:230,

                 from sketch\culfw-esp8266.ino.cpp:1:

C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\cc1100/cc1100.h:33:14: error: expected unqualified-id before 'void'

   33 |  void assert(void);

      |              ^~~~

C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\cc1100/cc1100.h:33:14: error: expected ')' before 'void'

C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\cc1100/cc1100.h:33:7: note: to match this '('

   33 |  void assert(void);

      |       ^~~~~~

C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\cc1100/cc1100.h:33:14: error: expected ')' before 'void'

   33 |  void assert(void);

      |              ^~~~

C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\cc1100/cc1100.h:33:7: note: to match this '('

   33 |  void assert(void);

      |       ^~~~~~

In file included from c:\users\roman\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\assert.h:10,

                 from c:\users\roman\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\sys\reent.h:503,

                 from c:\users\roman\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\stdlib.h:18,

                 from c:\users\roman\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\cstdlib:75,

                 from c:\users\roman\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\stdlib.h:36,

                 from C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:27,

                 from sketch\culfw-esp8266.ino.cpp:1:

C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\cc1100/cc1100.h:33:7: error: expected unqualified-id before ')' token

   33 |  void assert(void);

      |       ^~~~~~

C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\cc1100/cc1100.h:33:7: error: expected unqualified-id before ')' token

   33 |  void assert(void);

      |       ^~~~~~

In file included from C:\Users\Roman\Documents\Arduino\culfw.esp8266-CUL_V3_compatibility\culfw-esp8266\culfw-esp8266.ino:39:

C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\board/board.h:348:36: error: non-local lambda expression cannot have a capture-default

  348 |  #define lambda(obj, func)        [&](char *data) { obj . func (data); }

      |                                    ^

[... 341 lines cut ...]

culfw-esp8266:222:1: error: cannot convert '<lambda(char*)>' to 'void (*)(char*)' in initialization

culfw-esp8266:222:1: error: cannot convert '<lambda(char*)>' to 'void (*)(char*)' in initialization

culfw-esp8266:222:1: error: cannot convert '<lambda(char*)>' to 'void (*)(char*)' in initialization

culfw-esp8266:222:1: error: cannot convert '<lambda(char*)>' to 'void (*)(char*)' in initialization

Mehrere Bibliotheken wurden für "WiFiUdp.h" gefunden

 Benutzt: C:\Users\Roman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi

 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

'Memory' was not declared in this scope

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`
Gruß
Roman

romakrau, 2021-12-23:
Hallo Manfred,
habe die Arduino-IDE neu aufgesetzt (Version 1.8.19). Seitdem läuft's. Täte mir leid wenn ich Dich umsonst bemüht hätte. Was muss ich beachten, ausser HAS_433 wenn ich einen CC1100 mit 433 Mhz betreiben will.
Reicht das auskommentieren um auf 868 zu wechseln. Danke für Deine Mühen.
Schöne Weihnachtstage 
Roman

Man-fred, 2021-12-24:
Hallo Roman,
freut mich wenns läuft. Ich hatte noch Probleme den CC1101 anzusprechen, aber dann liegt es wahrscheinlich an meinem Chip. Wo gibt es eigentlich das HAS_433? Ich habe für mich nur eine Version und habe für die 433 MHz - Version per Register die Frequenz auf 433 eingestellt. Läuft stabil seit langer Zeit mit den Baumarkt-Sendern und Empfängern. 

Einstellung bei mir: CUL_ESP433 ccconf => freq:433.920MHz bWidth:325KHz rAmpl:42dB sens:4dB

Frohe Weihnachtstage
Manfred