Man-fred/culfw.esp8266 #15: MIS0 Pin nicht vorhanden

Comotio82, 2023-01-17:
Hallo, danke für das tolle Projekt.
Leider hat mein Board kein MIS0 Pin
Ich habe folgende:

GND
GOD2
GOD1
MOSI
SCK
CSN
GOD0
VCC

Kann ich das Board irgendwie verwenden?
Vielen Dank im vorraus!

[image]

Man-fred, 2023-01-17:
Hallo, 
ich kenne das Modul nicht, aber einen Hinweis habe ich gefunden: https://forum.fhem.de/index.php?topic=53559.0
Zitat: "... das Datenblatt vom CC1101 sagt aus, dass GDO1 mit SO doppelt belegt ist. Das ist eine verkürzte Bezeichnung für MISO. Alles unter der Voraussetzung, dass auf deinem Modul wirklich ein CC1101 ist."
Ich hoffe, das hilft.
Viele Grüße

Comotio82, 2023-01-18:
Vielen Dank dafür. Ich habe das Mal versucht, klappt aber irgendwie nicht.
Ich werde mir das Modul aus deiner Anleitung bestellen.
Liebe Grüße